dangreco/threedy #3: Option to add camera view

sebasptsch, 2021-01-28:
Replace the Printing Animation with the camera when available using a home-assistant camera entity.

dangreco, 2021-01-28:
I like the idea for this.
I'm thinking a good functionality would be to be able to switch between animation/camera view with a click/tap on the animation, providing the user with a default-camera or default-animation option in the config. 
Another possibility would be a button at the bottom to pop open the video feed in a modal. 
Thoughts?

sebasptsch, 2021-01-28:
Sounds good. Would you set either the camera or the animation as a default then?

dangreco, 2021-01-28:
I think the default would be camera if you specify a camera. 
Also, since the view is constrained to ~50% of the cards width, the feed may be cropped. A solution to this would be a click opens up a modal with the proper aspect ratio. Instead of clicking to switch between views, I could implement a swiping/scrolling feature between the two.

Codename-11, 2021-01-30:
I agree with @dangreco regarding the behavior. I think the animation itself is a staple feature of the card, but having a live view accessible via the card would be quite nice as well!

dangreco, 2021-01-30:
I'm ordering a webcam for my OctoPrint server so I can rapidly push out this feature as soon as possible while keeping quality a priority. From what I gather, video feeds are accessible by either a stream or MJPEG uri -- these seem relatively easy to implement. 

As far as how we would switch between views, I feel that my previous idea of swiping is not the way to go --- this would hinder workflow for mouse users. I'm going to instead refactor some of my code to add a camera button to the "header" of the card -- in the same space as the power and light icons. In this case, the previous idea of a pop-up modal will be sufficient.

Thoughts?

KTheMan, 2021-02-02:
@dangreco I actually have my octoprint webcam set up as a camera in HA. I wonder if it would make sense to just have the user just add a camera to their config and be able to pick it from a dropdown during card configuration.

MechEng70, 2021-02-04:
> @dangreco I actually have my octoprint webcam set up as a camera in HA. I wonder if it would make sense to just have the user just add a camera to their config and be able to pick it from a dropdown during card configuration.

completely agree with this approach.

dangreco, 2021-02-23:
This has officially been implemented in [the latest release](https://github.com/dangreco/threedy/releases/tag/2021.0.2). Thanks so much for your input!!

closed 2021-02-23 by dangreco