dangreco/threedy #10: Theme is broken

MechEng70, 2021-01-29:
downloaded most recent file: 2021.0.a1.2

Theme is still broken when using this resource.  

Configuration:

```
####################################################
#                    OCTOPRINT                     #
#################################################### 
octoprint:
  - host: 172.16.5.240
    api_key: !secret octoprint_api_key
    name: Creality CR-6 SE
    number_of_tools: 1
    sensors:
      monitored_conditions:
        - 'Current State'
        - 'Job Percentage'
        - 'Temperatures'
        - 'Time Elapsed'
        - 'Time Remaining'
```

Card:
```
      - type: 'custom:threedy-card'
        base_entity: sensor.creality_cr_6_se
        name: Creality CR-6e
        printer_type: I3
        monitored:
          - Status
          - ETA
          - Elapsed
          - Hotend
          - Bed
        theme: Default
        font: Roboto
        scale: 0.5
        round: false
```

When Bed or Hotend is removed, the card appears.

dangreco, 2021-01-29:
Could you reply to this with the console logs? Thanks!

MechEng70, 2021-01-29:
It doesn't like the monitored  state 'Bed', but it does like the state 'bed'.

[myplace-1611944845471.log](https://github.com/dangreco/threedy/files/5895580/myplace-1611944845471.log)

dangreco, 2021-01-30:
I've been working on some stability changes within the card. By the next release these issues should hopefully be fixed! I'm leaving this thread open so that we can continue to diagnose what exactly is the issue.

MechEng70, 2021-01-30:
Let me know if you need me to test something....  Thanks.

dangreco, 2021-02-01:
I think I fixed it! I will be doing another release tonight with more features including this bug fix.

MechEng70, 2021-02-02:
Can confirm that the card does not break the theme.  However, the "Bed" temp is not understood.

[image]

dangreco, 2021-02-02:
Make sure you have this in your Home Assistant configuration.yaml:

```yaml
octoprint:
  .
  .
  bed: true
  .
  .  
```

Thanks!

MechEng70, 2021-02-02:
Done.  Thanks!

closed 2021-02-02 by MechEng70