dangreco/threedy #11: HACS

alnavasa, 2021-01-29:
Waiting until it is HACS installable/updatable, Will this be one day?

Jan-Ka, 2021-01-30:
`hacs.json` is done, if the next release has files with the correct name (hacs wants https://github.com/dangreco/threedy/blob/a516447bf721f792486ad28b5a74797d10373df3/hacs.json#L5 but gets `bundle.js`) it should work.

Also the Repository needs Topics as per https://hacs.xyz/docs/publish/start#topics

Ping @dangreco

dangreco, 2021-01-30:
I've added the topics to the repo! The next release will indeed be following the ```threedy-card.js``` naming scheme. As far as when the next release is, I'm trying to ship out the graphical card configuration element as soon as possible. I've been testing multiple react component libraries trying to find one that is lightweight and compatible with the shadow DOM. ETA for the next release is 11:59PM (EST) on Sunday, Jan 31st. Stay tuned!!

Jan-Ka, 2021-01-31:
Thanks for the quick work!

Jan-Ka, 2021-02-01:
Ping @dangreco 

Found something else :(

My test for https://github.com/dangreco/threedy/pull/14 worked because I hadn't marked my test-release as pre-release.
As far as my tests show, you need at least one "real" release to be able to load the repository in HACS (at which point you can select "BETA" in the Install Modal)

[image]

https://github.com/Jan-Ka/threedy/releases/tag/2021.0.b1b is my test pre-release
https://github.com/Jan-Ka/threedy/releases/tag/2021.0.b1 is the previous test release

If I swap the first release to pre-release, I won't be able to add it again. Seems like a bug in HACS (or at least something that the user should be able to force through).

You could just release the first release as "initial release" as that would also have a file of the correct name. Just a suggestion.
Otherwise, we just have to wait until you make your first release.

dangreco, 2021-02-01:
I'm going to push out the first stable release asap -- I want to quickly implement a full graphical config for all options before release. Glad to know it works otherwise though!

dannytsang, 2021-02-03:
Managed to add this via HACS. The base git clone url doesn't seem to work until I used the links above: https://github.com/Jan-Ka/threedy/releases/tag/2021.0.b1b so can I suggest if I am adding the repo wrong or add it to the README?

apoc4lyps, 2021-02-04:
I think it's because there are only pre-release tags. @dangreco is there any plan for a release in the future?
Btw, thanks for the work, great card.

dangreco, 2021-02-09:
> I think it's because there are only pre-release tags. @dangreco is there any plan for a release in the future?
> Btw, thanks for the work, great card.

Yep, release coming soon! Been swamped with uni course work but gonna get around to finishing the initial release by this next weekend.

dangreco, 2021-02-18:
Should be available via HACS now that there is a release!

closed 2021-02-18 by alnavasa

luisleonardo, 2021-02-18:
I can't find it on HACS. I've searched for threedy, 3D and printer and no matching results.

Jan-Ka, 2021-02-18:
@luisleonardo it is not published to HACS, but you can add it as custom repository using this repo's URL now.

as per https://hacs.xyz/docs/publish/include
> Only the owner of the repository or a major contributor to it can submit a PR to have it included as a default.

luisleonardo, 2021-02-18:
And what's the correct link? Because the repo's URL gave me this message:
Repostitory structure for 2021.0.1 is not compliant

apoc4lyps, 2021-02-18:
I tried with https://github.com/dangreco/threedy and it works as expected.

luisleonardo, 2021-02-18:
I restarted HA and now it works. Thanks.

dannytsang, 2021-02-18:
I was able to add it successfully. Thanks @dangreco !