dangreco/threedy #6: Xo[c] is undefined in console and dashboard empty

ymarkus, 2021-01-29:
When I add a threedy card to my dashboard, that dashboard crashes and stays empty even if I add different cards to it.

My config (I've tried all monitors, all error out):
```
type: 'custom:threedy-card'
base_entity: sensor.monoprice_maker_select_plus
name: Monoprice Maker Select Plus
printer_type: I3
monitored:
  - Status
```

octoprint config (I've tried with & without the `sensors` segment):
```
octoprint:
  host: [..]
  api_key: [...]
  name: 'Monoprice Maker Select Plus'
  port: 443
  ssl: true
  bed: true
  number_of_tools: 1
  sensors:
    monitored_conditions:
      - 'Current State'
      - 'Temperatures'
      - 'Job Percentage'
      - 'Time Elapsed'
      - 'Time Remaining'
```

Console output: 
```
TypeError: Xo[c] is undefined
    as bundle.js:2
    to bundle.js:2
    Ns bundle.js:2
    Yu bundle.js:2
    gu bundle.js:2
    vu bundle.js:2
    mu bundle.js:2
    Ba bundle.js:2
    unstable_runWithPriority bundle.js:2
    Na bundle.js:2
    Ba bundle.js:2
    Va bundle.js:2
    su bundle.js:2
    Uu bundle.js:2
    Qu bundle.js:2
    render bundle.js:2
    value bundle.js:2
    set bundle.js:2
    value chunk.76c6271b07439f0dec52.js:2040
    _cards chunk.76c6271b07439f0dec52.js:2040
    value chunk.76c6271b07439f0dec52.js:2040
    value chunk.76c6271b07439f0dec52.js:2040
    performUpdate app.924deda6.js:10297
    _enqueueUpdate app.924deda6.js:10297
    requestUpdateInternal app.924deda6.js:10297
    initialize app.924deda6.js:10297
    C app.924deda6.js:10297
    i chunk.76c6271b07439f0dec52.js:2040
    value chunk.76c6271b07439f0dec52.js:2827
    value chunk.76c6271b07439f0dec52.js:2827
    setTimeout handler*i/< chunk.8bfaee99df77cd7cc01f.js:1
    requestAnimationFrame chunk.af43f6ec356055cbfe65.js:1
    requestAnimationFrame chunk.af43f6ec356055cbfe65.js:1
    i chunk.8bfaee99df77cd7cc01f.js:1
    value chunk.76c6271b07439f0dec52.js:2827
    performUpdate app.924deda6.js:10297
[... 201 lines cut ...]
    render app.924deda6.js:10297
    update app.924deda6.js:10297
    performUpdate app.924deda6.js:10297
    _enqueueUpdate app.924deda6.js:10297
    AsyncFunctionNext self-hosted:674
    (Async: async)
    requestUpdateInternal app.924deda6.js:10297
    set app.924deda6.js:10297
    value app.924deda6.js:43637
    value app.924deda6.js:39363
    performUpdate app.924deda6.js:10297
    _enqueueUpdate app.924deda6.js:10297
    AsyncFunctionNext self-hosted:674
    (Async: async)
    requestUpdateInternal app.924deda6.js:10297
    set app.924deda6.js:10297
    commit app.924deda6.js:10297
    commit app.924deda6.js:10297
    update app.924deda6.js:10297
    __commitTemplateResult app.924deda6.js:10297
    commit app.924deda6.js:10297
    a app.924deda6.js:10297
    render app.924deda6.js:10297
    update app.924deda6.js:10297
```

dangreco, 2021-01-29:
Good catch!!
I just fixed it and created a [new release](https://github.com/dangreco/threedy/releases/tag/2021.0.a1.2).
Thanks!

closed 2021-01-29 by dangreco

ymarkus, 2021-01-29:
Wow, that was quick! Thank you very much.