dangreco/threedy #8: Card-Mod style issue

Freshhat, 2021-01-29:
Hi, first of all great work on that card. 

Already integrated it into my overall overview into Lovelace. But I can't figure out how to apply different styles using card-mod to fit perfectly into my theme.

I just wanted to remove the box-shadow:

[image]

```              card:
                type: 'custom:threedy-card'
                style: |
                  ha-card: {
                    box-shadow: none !important;
                  }
                base_entity: sensor.ender_3_pro
                name: Ender 3 Pro
                printer_type: I3
                monitored:
                  - Status
                  - ETA
                  - Elapsed
                  - Hotend
                  - Bed
                theme: Default
                font: Roboto
                scale: 0.5
                use_24hr: true
                round: false
```
If i remove it directly under the element.style {} block in my Browser then it's working...
Any advise on that?

dangreco, 2021-01-29:
It seems to be an issue with how card-mod functions -- it tries to attach itself to the shadow DOM of the element, but my element doesn't have a shadow DOM to begin with. 
I'm working on wrapping the card in a shadow DOM to fix this.

dangreco, 2021-01-29:
Upon further research, the ```lovelace-card-mod``` readme says the following:
[image]

I attempted this workaround with my card and it works. For your situation, the style would look like: 

```css
threedy-card > div {
    box-shadow: none !important;
}
```

Hopefully this helps!

closed 2021-01-29 by dangreco