dangreco/threedy #9: FR: smarplug integration

giuliandenicola1, 2021-01-29:
Hi love the card! Can you add the possibility to show my kwh/watt from my smartplug?

And maybe some buttons like: print/pause/stop shutdown/reboot octoprint?

dangreco, 2021-01-29:
Hey! Thanks for the support.

The OctoPrint control buttons sound like a great idea, however the OctoPrint integration that Home Assistant uses is only for monitoring OctoPrint, not controlling (yet). I could try write an implementation that either uses a configured API key/URI or from Home Assistant's configuration, and then use that to make REST API requests. However, at that point I'd be using two different methods of interacting with OctoPrint and it would almost make more sense to abandon using the Home Assistant entities, which in turn defeats the whole purpose of using Home Assistant altogether. I'm unfortunately going to have to put this feature on hold for the time being.

Regarding the kwh/watt feature: while this is totally doable, I'm a bit hesitant to start extending functionality to things not directly associated with the printer (i.e., a power switch for the printer is OK, but something like this a bit niche and pertains to your smartplug specifically.) The rows of statistics are meant to represent only those statistics associated with OctoPrint, so I don't think it has a place for it there.

alnavasa, 2021-01-29:
I would like the possibility to turn on and off from a tasmota/sonoff already configured switch from Home Assistant
Thanks!

dangreco, 2021-01-29:
> I would like the possibility to turn on and off from a tasmota/sonoff already configured switch from Home Assistant
> Thanks!

This is already a feature!

alnavasa, 2021-01-29:
> > I would like the possibility to turn on and off from a tasmota/sonoff already configured switch from Home Assistant
> > Thanks!
> 
> This is already a feature!

Great then! thanks!

closed 2021-01-30 by dangreco