dangreco/threedy #15: Temperature Units

MechEng70, 2021-02-02:
The units for HA are in F and therefore show up on the card as F.  Is there a toggle that allows to convert and display units in either format?

[image]

dangreco, 2021-02-02:
I can implement this! ETA is initial release (Sunday Feb. 6)

dangreco, 2021-02-03:
Added in [my most recent commit](https://github.com/dangreco/threedy/commit/b16bc2b66707e13c9362a5a4b18c8ed046e194c2). Read the README for usage. Closing this issue -- expect a release Sunday as mentioned earlier.

Thanks!

closed 2021-02-03 by dangreco

MechEng70, 2021-02-08:
Hi @dangreco, what was the timing on the release?  Would like to test it.