dangreco/threedy #16: power_entity greyed out

konzolec, 2021-02-03:
Cannot get power_entity to work, the icon is greyed out.

Current configuration:

```
type: 'custom:threedy-card'
base_entity: sensor.ender_3
name: Ender 3
printer_type: I3
monitored:
- Status
- ETA
- Elapsed
- Hotend
- Bed
power_entity: switch.sonoff_basic_3d
theme: Default
font: Roboto
scale: 1
round: false
```

Entity ID in HomeAssistant`switch.sonoff_basic_3d`

`switch.sonoff_basic_3d `- Is a sonoff basic powercord

Any idea?

mikepowell, 2021-02-08:
I'm seeing the same with the `light_entity` attribute as well.
```yaml
  - type: 'custom:threedy-card'
    base_entity: sensor.ender_3_v2
    name: Ender 3 V2
    printer_type: I3
    monitored:
      - Status
      - ETA
      - Elapsed
      - Hotend
      - Bed
    theme: Default
    font: Roboto
    scale: 1
    round: true
    light_entity: light.basement_desk_lamp
```
I've confirmed the entity ID is correct, and works in other parts of home assistant UI.

Codename-11, 2021-02-09:
I'm seeing the same for both light and power.

Codename-11, 2021-02-18:
> 
> 
> I'm seeing the same for both light and power.

Fixed with new release it seems! Great work! Thank you very much for the beautiful card man! @dangreco

dangreco, 2021-02-18:
> > I'm seeing the same for both light and power.
> 
> Fixed with new release it seems! Great work! Thank you very much for the beautiful card man! @dangreco

No problem!! Thank you!

closed 2021-02-18 by dangreco