dangreco/threedy #18: Animation not showing on homeassistant ios app

aee74, 2021-02-15:
The animation works perfectly on my pc. But on the IOS app it's empty. even after clearing the frontend cache.

RobertD502, 2021-02-17:
To add to this, it also doesn't seem to be working in Safari on MacOS. My assumption is that this is a webkit compatibility issue.

dangreco, 2021-02-18:
This is a strange issue. The library I'm using for the animation, [Framer Motion](https://github.com/framer/motion) seems to support all browser variants. I have tested this (with success) on a webkit browser, but perhaps Apple's Safari does things differently. I'm going to borrow some Apple devices to test this on.

aee74, 2021-02-18:
[image]

aee74, 2021-02-18:
This with latest version on homeassistant ios app

RobertD502, 2021-02-18:
@dangreco I’ll investigate it with Dev tools on my Mac and iPhone in the AM and will report back any useful info that I find.

RobertD502, 2021-02-18:
@dangreco For some reason Safari on MacOS started displaying the card correctly. However, the card is throwing up the warning seen in the attached image. The warning appears whenever I load a lovelace dashboard that contains the card and also repeats itself every time I expand/collapse the card when the status of the printer is idle (when the card icon is blue).

[image]

iOS Companion App:

Your card is completely fine. This is purely a user error- in order for the card to work properly, iOS companion app users need to "Reset the frontend cache". Once that is done, the card works. The attached image is of the card viewed within the iOS companion app. It may be useful to add this to the instructions to avoid further issues of the same nature being opened. In order to reset the frontend cache, iOS users need to click on "App Configuration" in the sidebar from within the iOS Companion App --> then click on "Reset frontend cache" located at the bottom of the page. I do want to note that while debugging the card from within Mobile Safari on my iPhone, I received the same error as seen in the screenshot above- this doesn't cause the card to not work, so, it isn't really a pressing issue.

[image]

aee74, 2021-02-18:
I did a frontend reset. And still no animation
[image]

aee74, 2021-02-18:
[image]

RobertD502, 2021-02-18:
@aee74 I’m not sure if I’m the only one, but none of your uploaded images are showing.

aee74, 2021-02-18:
I’m also not seeing them. Try again
[image]
[image]

aee74, 2021-02-18:
I can see the image when idle but not while printing

[image]

RobertD502, 2021-02-18:
@aee74 Did you force quit the HA companion app after resetting the frontend cache?

dangreco, 2021-02-18:
> @dangreco For some reason Safari on MacOS started displaying the card correctly. However, the card is throwing up the warning seen in the attached image. The warning appears whenever I load a [...]

Digging around in the webpacked code revealed that the function `e.match` is part of the `framer-motion` library (or to my best judgement). I'll see if I can find a workaround by testing on Safari myself today + looking at the source code for `framer-motion`.

aee74, 2021-02-19:
> @aee74 Did you force quit the HA companion app after resetting the frontend cache?

Yes I did