dangreco/threedy #19: Problems installing card

Hobby4life, 2021-02-16:
I have copied threedy-card.js to folder /www/threedy-card.js

In Lovalace resources I have added different resource urls

/local/threedy-card.js
/www/thready-card.js
/local/www/threedy-card.js

Nothing works with these urls and the example configuration..

My printer is stated as: sensor.octoprint_ender3_state

I dont know where to look anymore.

MechEng70, 2021-02-16:
Where are you adding the lovelace resources?

Here is my set up:
card is located in /config/www/community/octopi-card/threedy-card.js

Added a resource by going to Configuration -> Lovelave Dashboards -> Resources and added:
 /hacsfiles/octopi-card/threedy-card.js as javascript

Then rebooted.

Hobby4life, 2021-02-17:
It is working!

Thanks to your setup :-)

This should be in the wiki manual.

"If you place the js file here, then use this path in resources.."

MechEng70, 2021-02-17:
Please go ahead and close this issue :)

closed 2021-02-17 by Hobby4life