dangreco/threedy #22: ETA/Elapsed jumping all over

thermoscookies, 2021-02-28:
Running version 2021.0.3

My ETA and Elapsed time are all over the place as shown in the GIF below.

[image]

Here is the history for both, they look "normal".
[image]
[image]

and my configs:
`type: 'custom:threedy-card'
base_entity: sensor.octoprint
name: Ultimaker 2+
printer_type: I3
monitored:
  - Status
  - Hotend
  - Bed
  - ETA
  - Elapsed
theme: Default

octoprint:
  host: 192.168.10.65
  api_key: 02E38742B0224BE488FB84CB7450537B
  bed: true
  number_of_tools: 1
  sensors:
    monitored_conditions:
      - "Current State"
      - "Job Percentage"
      - "Temperatures"
      - "Time Elapsed"
      - "Time Remaining"
'

Please let me know if there is something else I should check or change. Thank you

cmccambridge, 2021-02-28:
Possibly relevant observation - My configuration is `use_mqtt: True`, and in the Templates editor, the format of my print time sensor is `h:mm:ss`. (Ditto `_print_time_left`)
```
{{ states.sensor.prusa_mk3s_print_time.state }}
0:38:32
{{ states.sensor.prusa_mk3s_print_time_left.state }}
0:00:00
```

I don't know JavaScript or `moment` in particular but wondering whether that makes this conversion fail?
```
round ? moment.duration(time, "seconds").humanize()
```

dangreco, 2021-02-28:
> Possibly relevant observation - My configuration is `use_mqtt: True`, and in the Templates editor, the format of my print time sensor is `h:mm:ss`. (Ditto `_print_time_left`)
> ```
> {{ [...]

So the way I implemented the time entities was using the default data coming in from OctoPrint, which is just a number representing the seconds of some duration (e.g. 125 = 2m 5s). I didn't implement parsing templated time entities. What I will be doing is implementing custom monitored conditions, where you can use your templated entity directly instead of the ETA condition.

dangreco, 2021-02-28:
> Running version 2021.0.3
>
> My ETA and Elapsed time are all over the place as shown in the GIF below.
>
> [image]
>
>
> Here is the history for both, they look "normal".
> [image]
> [image]
>
> and my configs: [...]

This is probably just I bug I didn't consider while converting everything to typescript / cleaning up the code. I'll take a look at it today!!

cmccambridge, 2021-02-28:
And for completeness, the underlying MQTT message that the home assistant auto-discovered sensors are consuming:
```
{"_timestamp": 1614494674, "offsets": {}, "state": {"text": "Operational", "flags": {"cancelling": false, "paused": false, "operational": true, "pausing": false, "printing": false, "resuming": false, "sdReady": true, "error": false, "ready": true, "finishing": false, "closedOrError": false}}, "job": {"estimatedPrintTimeFormatted": null, "averagePrintTime": 2312.8079525809735, "lastPrintTime": 2312.8079525809735, "user": "pi", "file": {"origin": "local", "name": "Adapter_v3_0.2mm_PETG_MK3S_33m.gcode", "date": 1614492357, "path": "Adapter_v3_0.2mm_PETG_MK3S_33m.gcode", "display": "Adapter_v3_0.2mm_PETG_MK3S_33m.gcode", "size": 1180153}, "estimatedPrintTime": null, "filament": null}, "currentZ": 48.0, "progress": {"completion": 100.0, "filepos": 1180153, "printTimeFormatted": "0:38:32", "printTimeLeftFormatted": "0:00:00", "printTimeLeft": 0, "printTime": 2312, "printTimeLeftOrigin": null}, "resends": {"count": 0, "ratio": 0, "transmitted": 40820}}
```

where there *is* a raw seconds value.... but the auto-discovery is picking up the formatted one instead :-/ (noting the `printTimeFormatted`)
```
{"pl_avail": "connected", "name": "Prusa MK3S Print Time", "uniq_id": "x_PRINTING_T", "stat_t": "~hass/printing", "pl_not_avail": "disconnected", "avty_t": "~mqtt", "device": {"mf": "Prusa", "sw": "HomeAssistant Discovery for OctoPrint 3.2.1", "ids": "x", "mdl": "Prusa i3 MK3S", "name": "Prusa MK3S"}, "ic": "mdi:clock-start", "val_tpl": "{{value_json.progress.printTimeFormatted}}", "~": "octoPrint/"}
```

cmccambridge, 2021-02-28:
Oops, raced your reply there... but in other words- not *my* templated sensor, per se - just using the template to dump its raw content. Rather, this is the default format that any user will get when setting up the octoprint MQTT-based discovery plugin. Arguably a bug/less desirable feature from that plugin's side, actually... it would be easier to manipulate if they put the raw seconds value in that sensor.

deosrc, 2021-03-10:
Looks like changing the round time setting to "Yes" causes the value to stop jumping about but it increases exponentially instead. Unfortunately I don't think I know enough React or TypeScript to see anything in the code which may be causing this.

dangreco, 2021-03-10:
Fixed in [the latest release!](https://github.com/dangreco/threedy/releases/tag/2021.0.4)

Basically, JS was somehow treating the time and direction as strings and concatenating them hahaha. Explains a lot!