dangreco/threedy #24: Temperature units still show F after converting to C

chrisron95, 2021-03-01:
I see issue #15 addressed the conversion, which is working correctly, however looks like the frontend is still displaying F even after properly converting to C

[image]

dangreco, 2021-03-10:
Could you attach the raw config for your card? Thanks!

chrisron95, 2021-03-10:
Sure, here's straight from the code editor:
```
type: 'custom:threedy-card'
base_entity: sensor.octoprint
monitored:
  - Status
  - Hotend
  - Bed
  - ETA
  - Remaining
  - Elapsed
printer_type: I3
name: Maker Select Plus
scale: 0.5
round_time: false
temperature_unit: C
round_temperature: true
camera_entity: camera.octoprint
theme: Default
power_entity: switch.3d_printer
```