dangreco/threedy #28: Remaining/elapsed resetting

JorgeMaTeixeira, 2021-04-03:
Remaining and elapsed time is constantly resetting values.
I´m using MQTT integration, and i can see that entity for elapsed or remaning is working fine, without any kind of resets.

https://user-images.githubusercontent.com/66649736/113476510-d32ca380-9473-11eb-8f98-804bd278643b.mp4

snakedm, 2021-04-03:
I have exactly the same problem. And I also use for MQTT integration.

Please fix this problem, because I like your card very much. It is very well done: simple, clean and beautiful.
But with this problem, I'm not able to use it, unfortunately!

Thanks very much!

dudu631, 2021-04-05:
Can confirm this! Have the same problem with mqtt integration.

JorgeMaTeixeira, 2021-04-21:
Any chance to solve this problem?

narfel, 2021-04-25:
Same here, just it says "A few seconds" ago instead of a number.

eloo, 2021-05-04:
i guess its a problem with the space in the line. 
when i'm going in the edit mode i see shortly the whole text but it then disappears again.

btw.. i'm also using the mqtt integration

dangreco, 2021-05-04:
Hey guys! Sorry for such a delay for such a critical issue: I had final exams and my laptop up and died out of nowhere. Once my computer arrives I'll get this fixed ASAP. Apologies again!!

moppedz, 2021-06-04:
Same problem here. Still resetting the displayed time values on mqtt data

AtomicLaser, 2021-08-12:
Any updates on this?  I have just installed this card, using MQTT and having the same issues with the timer resetting

tonysurma, 2021-09-26:
same here with MQTT.  I love this card otherwise.  Any update @dangreco? Or even a hint as to why it may be happening and maybe someone else can make a pull request to address? Thanks

AndreyShpilevoy, 2021-11-15:
Hi all, I added a fix for this issue.
If @dangreco will merge it - I think it could be helpful for a lot of people.

If you want - you can compile code from my branch by yourself and just replace the js file in the plugin folder.

Thank you.

dangreco, 2021-11-25:
@AndreyShpilevoy Thanks so much for finding and implementing this fix! I've been so sidetracked with other things lately, but I'm glad there's still an interest in this card. I'm working on the designs for a version 2.0 rewrite -- this is something I'll try to do once final exams/work is on break for the holidays.

closed 2021-11-25 by dangreco