dangreco/threedy #47: when clicked "ETA" card in lovelace shows "Cannot read properties of undefined (reading 'state')"

mutthunaveen, 2021-12-28:
Hi, 
Recently i ve installed HACKS and updated threedy repository also. Now when im configuring im getting error as below

Cannot read properties of undefined (reading 'state'), 
Status and Bed temperature works fine
when clicked ETA or some other parameter the whole card doesn't show up.

This is the error im getting

`
**> Cannot read properties of undefined (reading 'state')
> type: custom:threedy-card
> base_entity: sensor.octoprint
> monitored:
>   - Status
>   - Bed
>   - ETA
> printer_type: I3
> name: ENDER
**
`

Please support.

Typhoan, 2022-01-04:
I have also experienced this issue. 

I am on the docker version of HA.

rw377, 2022-01-17:
I am also getting this issue. The card works fine picking up the Prusa API / sensors set up in config, but it can only read 'Status' and 'Bed' when looking at the Octoprint integration sensors.  The sensors are all working though when looking at them through home assistant otherwise.

thermopesos, 2022-02-03:
Same issue here. I'm assuming Threedy is dead :/

rw377, 2022-02-03:
@thermopesos refer to this issue, you need to make a sensor w/ template and that resolved it for me: https://github.com/dangreco/threedy/issues/43

victorigualada, 2022-11-11:
Another solution would be to use the MQTT plugin in OctoPrint and use the MQTT integration instead of the HA one. 

You need to:
1. In OctoPrint install MQTT and HA Discovery Plugins.
2. In the card configuration set: `use_mqtt: true`

That solved the issue for me. 

Also MQTT is way faster reporting data than the HA integration (which uses REST calls)

e11en, 2023-03-14:
@mutthunaveen @Typhoan @rw377 @thermopesos  I've created a fork of this repo and fixed some issues like the state undefined stuff, check it out here: https://github.com/e11en/ha-threedy-card

sammyke007, 2023-12-23:
> @mutthunaveen @Typhoan @rw377 @thermopesos I've created a fork of this repo and fixed some issues like the state undefined stuff, check it out here: https://github.com/e11en/ha-threedy-card

Tnx @e11en !