dangreco/threedy #48: Not able to configure...e is undefined

sambeetm, 2022-01-28:
After installation, it put the .js file in the following path
root@debian:/usr/share/hassio/homeassistant/www/community/threedy# ls
threedy-card.js  threedy-card.js.gz

I copied the files back to following path
root@debian:/usr/share/hassio/homeassistant/www/hacsfiles/threedy# ls
threedy-card.js  threedy-card.js.gz

Still getting the error.

2022-01-28 14:57:58 ERROR (MainThread) [frontend.js.latest.202112291] http://192.168.1.66:8123/hacsfiles/threedy/threedy-card.js?hacstag=33288368520211125:2:602711 TypeError: e is undefined
2022-01-28 14:57:58 ERROR (MainThread) [frontend.js.latest.202112291] http://192.168.1.66:8123/hacsfiles/threedy/threedy-card.js?hacstag=33288368520211125:2:602711 TypeError: e is undefined
2022-01-28 14:57:58 ERROR (MainThread) [frontend.js.latest.202112291] http://192.168.1.66:8123/hacsfiles/threedy/threedy-card.js?hacstag=33288368520211125:2:602711 TypeError: e is undefined
2022-01-28 14:57:58 ERROR (MainThread) [frontend.js.latest.202112291] http://192.168.1.66:8123/hacsfiles/threedy/threedy-card.js?hacstag=33288368520211125:2:602711 TypeError: e is undefined

muslimsteel, 2022-02-10:
I am also getting this error and it does not display anything in the UI

nickwilkie, 2022-06-23:
I get this error when I try to use the card with any monitored property that has a time, like ETA or Remaining. Temperatures seem to work fine

Mavi222, 2022-07-01:
I've got the same problem, and as nick said, if I check anything with time (time remaining/elapsed/ETA), it doesn't work and says an error, but when I uncheck those, it shows the plugin again.

siparker, 2022-07-26:
I also have the same issue. This originally caused the page to disappear for me when editing. that was on HA version 2021.11.5
i updated to 2022.7.6 and i now see an error instead of a blank page.
One note i am not sure if it matters. in the docs for manual card config it says `base_entity — Take the beginning of one of the OctoPrint sensors of your printer. Example: for sensor.ender_3_v2_current_state it would be sensor_ender_3_v2.`

However the setup doesnt do this. for me it just has `base_entity: sensor.octoprint`
not sure if that is relevant or not.

victorigualada, 2022-11-11:
Refer to my comment in another issue:
https://github.com/dangreco/threedy/issues/47#issuecomment-1311958802

e11en, 2023-03-14:
@codeclinic @siparker @Mavi222 @nickwilkie @muslimsteel @sambeetm  I've created a fork of this repo and fixed some issues like the state undefined stuff, check it out here: https://github.com/e11en/ha-threedy-card