dangreco/threedy #49: display permanently

jybi89, 2022-02-13:
hello, thank for your job !

I would like to permanently display the card without having to click on it

[image]

permanently display : 
[image]

thx for your answer

jybi89, 2022-02-14:
i solved it :)

`type: custom:mod-card
style: |
  threedy-card div div div:nth-child(2) {
    height: auto !important;
    opacity: 1 !important;
  transform: none !important;
  }
card:
  type: custom:threedy-card
  base_entity: sensor.octoprint
  monitored: []
  printer_type: I3
  temperature_unit: C
  use_24hr: true
  power_entity: null
  light_entity: light.wled_ender
  name: ' '
  camera_entity: camera.octoprint
  theme: Default
`

theclue, 2023-01-10:
Perhaps I did something wrong, but this actually doesn't work to me:

```
type: custom:threedy-card
base_entity: sensor.ender3
card_mod:
  style: |
    threedy-card div div div:nth-child(2) {
    height: auto !important;
    opacity: 1 !important;
    transform: none !important;
    }
monitored:
  - Status
  - ETA
  - Remaining
  - Elapsed
printer_type: I3
name: Ender 3
use_mqtt: true
theme: Default
use_24hr: true
```

e11en, 2023-03-14:
@theclue  I've created a fork of this repo and added an option to alway display, check it out here: https://github.com/e11en/ha-threedy-card