dangreco/threedy #53: add option via HACS not working

roblatour, 2022-09-18:
I am using:
Home Assistant 2022.9.4
Supervisor 2022.08.6
Operating System 9.0
Frontend 20220907.2 - latest

I tried Method 1: HACS

Open HACS and navigate to Frontend Section
ok

Open the Overflow Menu (⋮) in the top right corner and click on Custom repositories
ok

Paste https://github.com/dangreco/threedy into the input field and select Lovelace from the dropdown
ok 

Click the Install Button on the highlighted Card titled threedy
there was no 'Install' button, but there was an 'Add' button, clicked that it seemed to work

However, after that, and after a restart I tried to add a new card to my Home Assistant Overview.  I expceted to see a card for Treedy but did not, nor did I see any new cards.

Maybe this is a bug, maybe I'm missing something?

briancmoses, 2022-09-25:
I had similar problems--it seems some of the documentation might need some attention.

I worked around this,  though.  Here's what worked for me:

1.  Open HACS
2. Select Frontend
3. Click the **EXPLORE & DOWNLOAD REPOSITORIES** button in the lower righthand corner.
4. Enter *threedy* in the Search for repository .
5. Click the Download button and then click Download again in the window that pops up.
6. Click Reload.

roblatour, 2022-09-25:
Thanks Brian, I'll give it a try but I've got a print running for the next few hours.

briancmoses, 2022-09-25:
I hope it helps!

My printer has been running the whole time I've been messing with Home Assistant and setting up this card,  I think you'll be safe.

But I also wouldn't want to goof up an hours-long print,  either!

roblatour, 2022-09-25:
Hi Brian, I gave it a try but I'm generally getting the same results.

In the HACS Front End window, I get a single card that looks like this:

[image]

and if I click on it a documentation page comes up.  It says I should configure it but not where this is done.   I assume on the Home Assistant Overview dashboard, as a manual card?  But if I use the example given it doesn't like it.  This what I get:
[image]

It may be I am missing a small but important step, if so I just can't figure out what it might be.

roblatour, 2022-09-27:
I feel I am getting closer, here is what I have now:

in Dashboard edit:
 
[image]

[image]

in the Dashboard:

[image]

Any ideas?

roblatour, 2022-09-28:
still more progress, but not fully working ...

type: custom:threedy-card
base_entity: sensor.octoprint
name: PRUSA MK3S
printer_type: I3
monitored:
  - Status
  - Hotend
  - Bed
theme: Default
font: Roboto
scale: 1
round: false
temperature_unit: C
use_24hr: false

However, I can't add Remaining:

Here it is without Remaining

[image]

Here it is with Remaining:

[image]

mavermc, 2022-10-11:
I also have gotten this exact issue, any Time based variables like ETA/ Elapsed or Remaining seems to break it.

killtux, 2022-12-03:
Solution 1: add the sensors the plugin is looking for as template:
https://github.com/dangreco/threedy/issues/43#issuecomment-962651048

Solution 2: use mqtt for sensors:
https://github.com/dangreco/threedy/issues/47#issuecomment-1311958802

chevyman142000, 2023-03-08:
I can't get threedy-card or mushroom-endity-card to work. They all say custom element doesn't exist. Any suggestions?